RomeHein/ESPecial #1: Request: Add action HTTP and HTTPS calls

sjfaustino, 2020-05-08:
One use case is posting the battery voltage from the ADC pin to Thingspeak.

RomeHein, 2020-05-09:
Yes indeed, amongst many others! It’s definitely the next thing being developed. I’m working on the i2c integration atm, this is a very big one. I think I’ll merge a first beta version of the i2c to implement http actions first. Because this can be implemented pretty quickly compare to i2c

RomeHein, 2020-05-12:
I've pushed on master something that should help you. I've implemented http call for get and post methods. At the moment I haven't done https yet. Also it's pretty basic, you can't for instance set a custom header in you action for post requests. You can only send json atm.
I've also added a useful functionality: wherever you can set text in action panel, you can dynamically call the value of a pin directly in your string/json.
For instance, let's say you want to send a json via a post request with some value from pin 2, you would do something like that:
```{ key1: "bla", key2: "blabla", valuePin2: ${2} }```
So the syntax is `${pinNumber}`
You can also call system informations with `${info}`
Let me know if you have suggestions or else, hope this will help.

closed 2020-05-15 by RomeHein