RomeHein/ESPecial #3: Bug: When you add a GPIO control, edit it and then delete it, the pin number isn't available anymore

sjfaustino, 2020-05-14:
As specified on the title, using your latest code.
If you restart the esp, the pin numbers show up again as available to be used.

RomeHein, 2020-05-14:
This is a front issue, should be ok to fix quickly. I’ll just update the available GPIO list just after any delete.
Thanks!

RomeHein, 2020-05-15:
Actually I can't reproduce the bug. Could you add more details?
What I've tried so far:
Add a Gpio -> edit it -> delete it -> try to add it again (without reloading nor restarting the board) -> found the same pin in the list 🤷‍♂️

sjfaustino, 2020-05-16:
Attached is a video of the bug (need to rename the zip to mp4 since github doesn't allow this extension.

Also, the invert state checkbox isn't reflected when you edit.

[gpiobug.zip](https://github.com/RomeHein/ESPECIAL/files/4638637/gpiobug.zip)

RomeHein, 2020-05-17:
I've pushed another try of the fix. Let me know if it worked for you. This might have come from server error when trying to get the new list of gpio just after a deletion. Meaning that it might have not happened everytime. I've fixed it by not making any api call anymore, but removing from the list directly in the front. This make me think that I really need to move the api to an async server. At the moment it's really slow because each call is synchronous.

sjfaustino, 2020-05-17:
Fixed. I can now select a pin after deleting it and the inverted state is reflected when I edit the control.

closed 2020-05-17 by sjfaustino