RomeHein/ESPecial #5: Sugestion: Check if new firmware available and optionally update automatically

sjfaustino, 2020-05-17:
Just saw the comments on the latest update regarding compilation difficulty for beginners if you move away from Arduino.
My sugestion is to have available a binary on github where you just flash it to the esp32 and allow the firmware to check if new firmware available and optionally update automatically or even revert to a previous version (show a list of available versions with a button to "go to old version" and "just update"?)

Examples of code: https://github.com/espressif/arduino-esp32/tree/master/libraries/Update and https://github.com/programmer131/ESP8266SelfUpdate

RomeHein, 2020-05-17:
Yes that would be really nice. That's actually on the wishlist since the beginning:
```Auto update: from remote server (why not a .ini build on this repo?)```
The only concern I have is when I make a change in the data structure. Any change might corrupts what is stored in the flash. So I need to make some sort of migration tool. Which can become tricky. 
I think a first step would be to give the possibility to export the current configuration as a json, and have the possibility to import a configuration (as a json file). So this would be very easy to save your current configuration before doing an update. This would also be very useful if you have multiple esp32 and want to export the same configuration through all of them.
That would also simplify the work on a migration tool.
But for now a simple auto update will do, I just need to keep this in mind. So this issue should not been closed until all these things are not implemented ;)

RomeHein, 2020-06-04:
Ok, so with the v0.2 of this repo, I've implemented a new way to update ESPecial. The webinterface will now fetch (via the esp32) versions available on this repo. This unlocked at the same time a pretty nice feature: installation via [ESPInstaller](https://github.com/RomeHein/ESPInstaller). This save a lot of trouble with dependencies installations and spiffs transfer.

closed 2020-06-04 by RomeHein