RomeHein/ESPecial #7: Bug: Analog read is defined incorrectly for pins

sjfaustino, 2020-05-17:
When you select a "Input mode" (should it not be "I/O mode"?), you can select Analog from the ist for pins which do not have that function and it doesn't show for pins that DO have the function.
I' ve compiled a list from https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html that might help:
`adc1 ch0 - Pin 32
adc1 ch1 - Pin 33
adc1 ch2 - Pin 34
adc1 ch3 - Pin 35
adc1 ch4 - Pin 36 - Conflict with HALL sensor 
adc1 ch5 - Pin 37
adc1 ch6 - Pin 38
adc1 ch7 - Pin 39 - Conflict with HALL sensor

adc2 ch0 - Pin 0 - Conflict with Wifi - Conflict with bootup
adc2 ch1 - Pin 2 - Conflict with Wifi - Conflict with bootup
adc2 ch2 - Pin 4 - Conflict with Wifi
adc2 ch3 - Pin 12 - Conflict with Wifi
adc2 ch4 - Pin 13 - Conflict with Wifi
adc2 ch5 - Pin 14 - Conflict with Wifi
adc2 ch6 - Pin 15 - Conflict with Wifi - Conflict with bootup
adc2 ch7 - Pin 25 - Conflict with Wifi
adc2 ch8 - Pin 26 - Conflict with Wifi
adc2 ch9 - Pin 27 - Conflict with Wifi
`

RomeHein, 2020-05-17:
Very nice, I'll add that asap. Thanks

RomeHein, 2020-05-17:
Ok... So another mistake from misunderstanding some documentations. I should not have call this "analog" but "Led Control" or maybe "PWM". Nothing to do with ADC... Sorry about that. I'll rename it. ADC looks pretty easy to implement with the arduino IDE though, at least with basic features. That's something I could do, but I think I'll finish off first the implementation of the I2C protocol, for the moment it's only possible to scan addresses. I'd like to be able at least to send data on specific registers.

RomeHein, 2020-06-15:
This issue has been addressed in the version 0.4 of ESPecial:
51b78ee887c1c82acb9e61fc87d3daebe25bde79
Important note: as ESPecial is mainly running on wifi, because of that, ADC2 won't work. So only gpios from 32 to 39 will be able to read analog input. Check expressif documentation on that point: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html

closed 2020-06-15 by RomeHein