RomeHein/ESPecial #16: Is it possible to add device ID to option selection?

happytm, 2021-02-04:
This is simplest Automation for ESP devices I have seen.

Is it possible to add device ID to option selection at front end? It will be very useful to control multiple ESP devices if they are identified by their preset device ID. It will be lot easier to run your excellent software on only one master ESP device and to control few of the slave ESP devices.

Thank you.

RomeHein, 2021-02-05:
Sorry I'm not sure to understand your demand.
You can already chose the dns name in the setting panel. This will give you the possibility to access your esp32 in that manner:
```
http://your-custom-name.local/
```
This way you can easily send http requests to the esp32 you are considering as slave.

happytm, 2021-02-05:
No I was not talking about dns name.

I was hoping you can add option selection drop box for device ID just above pin selection (shown below) for Device ID. The master ESP32 device then receives a command for particular pin on particular device and send MQTT message to that particular device.

[image]

Thanks.

RomeHein, 2021-02-06:
ok, if I understand correctly you are taslking about custom MQTT topic for each PIN?
I was planning to do something like that, and also the possibility to add MQTT automation: https://github.com/RomeHein/ESPecial/projects/1#card-48292658
This is something I'll developed soon.
I can see from your screenshoot that you are using an old version of ESPecial. Try to install the new version, it contains a tons of new features. If you do so, you'll certainly have to reset all you gpios settings.

happytm, 2021-02-06:
I am sorry.
Thank you.