RomeHein/ESPecial #17: Telegram function does not work

zilhos, 2021-07-22:
Good morning, really grateful for sharing your work is excellent
I could never make the telegram function work
Even if I put the universaltelegrambot repository that you mention directly it doesn't compile the ide.
Question ? Can you compile in a .bin file the version you use and all the libraries that work correctly for you?
In this way there would be no error in libraries or versions.
From this moment I thank you for your answer
Greetings from Argentina
[image]

[image]

RomeHein, 2021-07-26:
Hey zilhos!
Thanks for giving a try to Especial :)
I'll have a look to this issue today. I've done a lot of refactoring without checking Telegram features heavily. I might have broke that.
It's a bit annoying, cause I don't like this workflow: refactoring then manually checking/testing features. 
If only they were a way to automate tests. I might have to investigate [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/unit-tests.html) to avoid your type of issue in the future.

To answer to your question on a precompile .bin file, you can find the latest one in the folder "version" of this repo.
But apparently I haven't updated it lately... oops
I'll have a look at it and keep you updated ;)

RomeHein, 2021-10-09:
Hello!
Sorry for the delay to solve this issue. You might have gone for another solution since then.
But last commit solved the issue. UniversalTelegramBot has also merged my suggestions, which means we don't have to use my fork anymore but the official repo, in order to build the project.
Don't forget you can use [ESPInstaller](https://github.com/RomeHein/ESPInstaller) to install ESPecial without going through the hassle of installing all the dependencies.
New released version 0.8 allows you to also use ESP-cam, send picture to Telegram (and of course use automation to do so)
Enjoy!

closed 2021-10-09 by RomeHein

zilhos, 2021-11-01:
Excellent work! Thank you very much for sharing. I also appreciate your answer. I am already using version 8.0 is excellent and I have no problems with telegram, everything flows wonderfully. I expect many more improvements I think you are incredible programming. Thank you very much and cordial greetings