RomeHein/ESPecial #18: Not All ESP32 Boards Support Partition Scheme

RebbePod, 2021-11-07:
I have a ESP32 DOIT dev kit V1. It doesn't have a Partition Scheme option.

RomeHein, 2021-11-19:
Hey! Sorry for the late answer! 
I have the same esp32 board, the partition schema is an option you'll find in your Arduino IDE.
Have a look to the first section of this page: https://robotzero.one/arduino-ide-partitions/
Let me know if that does not answer your question.

RebbePod, 2021-11-19:
Thank you for reaching out better later then never:)

The partition scheme option wasn't showing up, what I did was selected a different board and flashed it. This is all in the Arduino IDE

RomeHein, 2021-11-19:
Ah yeah that's right. I don't know how it works, but I guess there is a conf files somewhere in arduino that describes what partition schemes are available for our boards. I would say for that one it's missing the minimal SPIFF partition 🤷🏻‍♂️ 
Maybe check if your board manager is up to date.

closed 2021-11-20 by RomeHein