RomeHein/ESPecial #19: telegram chat

josef2600, 2023-03-04:
hi, love your work and idea. 
i wanted to chat with you in telegram with the link you posted "https://t.me/joinchat/AAAAAET-x6q-63Q4ZmFGng" but i cant do anything. there is only one message in it and nothing. i was wondering if it is possible to chat with you? and is there an newer version? i read somewhere you said you are working on a new version and it is way better and bigger than this. when will it be finish? 
anyway, thank you. regards, Josef.

RomeHein, 2023-03-04:
Hi @josef2600 
Thanks for reaching out!
I created the telegram channel for people like you who need support 😄 
Just write your question/remark there and we can discuss further.
For the rest, I must confess I don't work much on the project anymore. It's a pretty big one, and I definitely need help to continue working on it. 
There is so much to do/improve ! The potential is pretty big, but I didn't get much traction from the community, so I lost a bit the interest.
It would be super nice if you could help of course 👍👍👍
Let's discuss this on the telegram channel

josef2600, 2023-03-05:
thanks. but as i said, i cant chat in your telegram channel. i think it is closed for chat. 
on the other hand, i understand you completely and i want to do alot with you, if you are willing. i cant (don't want!) to discus them in open. so please reach me via my email or Skype. i am waiting for your response. thank you.

josef2600, 2023-03-06:
i have made few changes to your project. https://drive.google.com/file/d/1g4rc94-JPN1sRTvgWR1YRB531Qj962da/view?usp=sharing
first i defragment the files and folders. i separated the main files from other library files. that was hard! 
then there was few things i had to fined. for example i wanted to remove touch capability because it doesn't work properly. i haven't tested it on new ep32's. but i have ordered them. i will receive them in few days. my hope is with esp32s3. and i will test c3 too. but in general i work touch pads with pic16 new architecture, as i said i am a genius in hardware department. they are perfect! if i may say! and the ADC doesn't work on all pins. specially it doesn't work on strap pins. the reason is obvious! so i tried to remove those too. (i made comment on the "PreferenceHandler.cpp" file for those changes. 
and i made a change for it to be able to work with rs485 protocol. one pin is needed for it. check the schematic. 
then i added the capability for having a nice clock. now it automatically reads the clock from ds1307 and if we have access to internet, it update the clock every 1 hour (i think?!). the numbers are changeable. this was the hardest. there are 4 main reason i added external clock ic, because the modules do not have crystal for clock, and VDD RTC is connected to VDD. then there is the external SRAM inside the clock chip that works with battery! and last is the accuracy of the internal clock. its really bad! but they say in the new chips they have fixed it. base on the datasheet, the accuracy of the internal clock for s3 is amazing! 
the last thing i fiddled with (as i remember!) was the psram. 
at the end, i found a problem. we have a leak in memory. more specifically, in psram. wen it does something and it is finished, it consume memory but doesn't release it all. captain we have a leak !! 
it was don few mount ago and i have a very bad memory. i don't know what is wrong with me in that department (at least). i have ADD that make me very frustrating to even live! in the past few years it is gotten worth. so i lose what i did or write in codes. 
anyway, i hope it picked your interest! and maybe gave you hope. and there is something i should say about users! i think most people as zombies. they don't think, they only consume and destroy everything they see. they believe what they read and see only what is in front of them. they do not search or look at it with thinking point of view, hell, they don't even have a point of view. only what they are fed. but those are lies and manipulation! don't worry, i am not a politic person! i only have a point of view and some times read history and research. anyway, hope it was helpful. 
but to answer your most important question, is it profitable? well, i hope so! that is why i am working, and so you do. without money, nothing can be don or go forward. i have few plans for it. i don't have lot of money, but i can make it and start it. if you are willing and have patience, i love to work with each other. hope i made you fill better.
regards, Josef.

RomeHein, 2023-03-06:
Hi @josef2600 
Thanks for your work, this is much appreciated 👍

I would suggest to open a pull request, this way we can easily discuss and review the changes you made. 
Here is a link that explain how to contribute on Github: https://git-scm.com/book/en/v2/GitHub-Contributing-to-a-Project
In brief, usually this is how we proceed in the software world:
- clone the repo on your machine
- create a new branch with an explicit name (usually about the features/bugs you are adding/fixing)
- make the changes you want.
- commit the changes to your branch
- open a PR (pull request) on github with your branch
- wait for the reviewer to check/validate your changes

In a PR, we try to keep it small, and around the same concern/feature/bug. If you have multiple things, open multiple PRs, each one with a simple/small feature or bug fix.

Once you have clean your work, we can discuss further the content of what you said in the PRs you'll create.

Just a couple things:

>it was don few mount ago and i have a very bad memory. i don't know what is wrong with me in that department (at least). i have ADD that make me very frustrating to even live! in the past few years it is gotten worth. so i lose what i did or write in codes.

I'm sorry to hear that, but I can't help you on this 😕 
In general I would say that any dev would forgot what they wrote after few weeks. To be honest, I don't remember this project at all. Even though I wrote 100% of the code and spent a lot of time on it. You problem is really common to everyone. If you don't maintain a memory, it will eventually be erased by some chemical process in your brain. So don't worry to much about it.

> anyway, i hope it picked your interest! and maybe gave you hope. and there is something i should say about users! i think most people as zombies. they don't think, they only consume and destroy [...]

This is very personal point of view. To be honest I don't really understand what this is about, and why it's here. I would suggest to avoid this kind of discussion in project like this one. They are absolutely not related. 
In the future, please review your comments and remove this kind of discussions before posting a PR/issue/comment. They don't add any values to the conversation.
I think this is a valid suggestion for any kind of open source projects you may be participating to 😉

josef2600, 2023-03-06:
thank you. i have edited my comment. 
i just found out the Jason lib in Arduino had a few major changes. i made the changes but haven't tested out. when i tested it, i will upload it somewhere! hopefully in GitHub as you mentioned. funny, as i remember, it made few warnings regarding I2C creation, but now i didn't get any warning. don't know why! but i just wanted to inform you and i will discuss it later properly. thanks again. and did you check your telegram channel? i can see your massage, but i cant write anything. i think you need to change some setting. about my inappropriate stories, sometimes I tend to get lost in my thoughts. sorry for that.
regards, Josef.