RomeHein/ESPecial #23: C3: panic attack!

josef2600, 2023-03-21:
`Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.`
now, it runs, works, and suddenly, panics! i setup 3 analog input pins and read them. if the an2 is over 0, it send a massage in serial port (i am using USB inside the chip as serial port). after awhile, it panics, then hangs, then reboot itself.

`Free heap:   151524
Used heap:   130524

[PIN] Gpio pin 2 state changed. Old: 133, new: 131
[EVENT] Sent GPIO event for pin 2. Old: 131, new: 131
[PIN] Gpio pin 4 state changed. Old: 117, new: 116
[AUTOMATION] Checking automation: l1
[AUTOMATION] Running automation: l1
[ACTION] Running action type: 3
pppppppppppppppppppppppppppppppppppppppppppppppppppppppppppppppp
[AUTO_LOOP] Checking time scheduled automations

Free memory: 151524
Telegram bot: 0
Mqtt server: 0

Free heap:   151524
Used heap:   130524

[PIN] Gpio pin 2 state changed. Old: 131, new: 128
[EVENT] Sent GPIO event for pin 2. Old: 128, new: 128
[PIN] Gpio pin 3 state changed. Old: 158, new: 155
[PIN] Gpio pin 4 state changed. Old: 116, new: 119
[AUTOMATION] Checking automation: l1
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4200b926  RA      : 0x4200b934  SP      : 0x3fcba390  GP      : 0x3fc90200  
TP      : 0x3fc967c0  T0      : 0x30363731  T1      : 0xb1800000  T2      : 0x203a6469  
S0/FP   : 0x3fcbacbc  S1      : 0x3fcbb820  A0      : 0xe007606a  A1      : 0x00000000  
A2      : 0x00000000  A3      : 0x0000b7a9  A4      : 0x0002b7a9  A5      : 0x00000058  
A6      : 0xfa000000  A7      : 0x00000003  S2      : 0x3fcbb82c  S3      : 0x3fca2140  
S4      : 0x3fcb0494  S5      : 0x00000005  S6      : 0x00000003  S7      : 0x00000001  
S8      : 0x00000001  S9      : 0x3fca217a  S10     : 0x3fc9a000  S11     : 0x00000004  
T3      : 0x6d6f7475  T4      : 0x61203a74  T5      : 0x6e657665  T6      : 0x0a0d3239  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0xe0076072  
MHARTID : 0x00000000  

Stack memory:
3fcba390: 0x3fcbb820 0x3fcbb850 0x3fc9a000 0x4200ba5a 0x4038b0f4 0x00000005 0x00000000 0x3fcbc5f8
3fcba3b0: 0x3fc99a50 0x3fc94000 0x3fca2141 0x3fcbc5f8 0x0000002f 0x0000002c 0x0c008400 0x87a20147
3fcba3d0: 0x3fc99a50 0x3fc94000 0x3fca2141 0x4200090a 0x00000000 0x3fc9a000 0x00000000 0x40388c52
3fcba3f0: 0x420218ea 0x420218b4 0x3fcba470 0x4038a804 0x3fc967c0 0x30363731 0x10000000 0x00000000
3fcba410: 0x00000033 0x00000033 0x3fcdfc10 0x00312d31 0x00000000 0x00000000 0x3fcdfc10 0x40388c6c
3fcba430: 0xa0000000 0x0000000a 0x3fc99acc 0x00000000 0x00000100 0x3fcba4bc 0x3fc99ad4 0x3fc99ae0
3fcba450: 0x3fc99ad4 0x3fcba4bc 0x00000100 0x00000000 0x3fc99acc 0x00000033 0x00000033 0x420218fc
3fcba470: 0x400391e6 0x00000001 0x3fc9a000 0x3fc9a000 0x3fc9a000 0x3fcba4bc 0x3fc9a000 0x3c0d18fc
3fcba490: 0x3fc9423c 0x3fcba528 0x3fcba4bc 0x4202223c 0x00000001 0xffffffff 0x3fcbc594 0x40383ca2
3fcba4b0: 0x3fc9a000 0x3c0d2000 0x3fcba528 0x4e49505b 0x7047205d 0x70206f69 0x34206e69 0x61747320
3fcba4d0: 0x63206574 0x676e6168 0x202e6465 0x87a20147 0x36313120 0x656e202c 0x31203a77 0x3c0d2000
3fcba4f0: 0x3c0d1000 0x3c0d1000 0x3fc9a000 0x3fc9a000 0x3fca2140 0x00000005 0x00000000 0x00000838
3fcba510: 0x3fc93d7c 0x00000001 0x3fca219c 0x42000e32 0x3fc9a000 0x3c0d2000 0x00000001 0x3fc94000
3fcba530: 0x00000077 0x3fc99a4c 0x3fca2140 0x42000fcc 0x00000000 0x3fca2140 0x3fc9a000 0x3fc99a40
3fcba550: 0x32312d32 0x00000038 0x3fc9a000 0x87a20147 0x3c0d1000 0x3c0d1000 0x3c0d1000 0x00000000
3fcba570: 0x00000000 0x00000000 0x00000000 0x00044dc0 0x3c0d1000 0x3c0d1000 0x3c0d1000 0x00000000
3fcba590: 0x3fc9a000 0x3fc9a000 0x3fc94000 0x4200120a 0x00000000 0x00000000 0x00000000 0x00000000
3fcba5b0: 0x00000000 0x00000000 0x00000000 0x87a20147 0x00000000 0x00000000 0x00000000 0x00000000
3fcba5d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x4038b086
3fcba5f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcba610: 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678 0x00000160 0xabba1234 0x00000154 0x3fcba240 0x0002a744
3fcba630: 0x3fc95a98 0x3fc9f53c 0x3fcba628 0x3fc95a90 0x00000019 0x10df3f58 0x84e61ad6 0x3fcba628
3fcba650: 0x00000000 0x00000000 0x3fcb7618 0x6f747561 0x6974616d 0x48006e6f 0x0010bd4a 0x00000000
3fcba670: 0x3fcba610 0x00000000 0x00000000 0x3fcbba90 0x4203021e 0x00000000 0x3fc9b044 0x3fc9b0ac
3fcba690: 0x3fc9b114 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x420ba78c
3fcba6b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcba6d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcba6f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcba710: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcba730: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcba750: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcba770: 0x00000000 0x00000000 0x2b000000 0xbaad5678 0x00000060 0xabba1234 0x00000054 0x3fcba78c

ELF file SHA256: ffc0333f2266536b

Rebooting...
ESP-ROM:esp32c3-api1-20210207
`
for now i am working on changing strings on air to a global register. again, please when programming for microcontrollers, never use C++ as it is the worst language for it. read a datasheet, any micro from last 25 years that still exist. if they support any high language, it is always C and never C++. there are many reasons for it. 
eg:
what is the difference between array and string?
in C, nothing! and that is logic answer. BUT
A "C" string:

`char *myString = "This is my string";`

is an array of characters. It is no different to an "array".

However, a "C++ String Object" is vastly different.

`String myString = "This is my string";`

That is an entire object with methods and properties which, while it makes programming easier, is not advised when working with microcontrollers. The memory usage when you're not doing much with it is only slightly more than a "C" string. However, a lot of the methods involve the allocation and deallocation (freeing) of memory, which leads to memory fragmentation.

"C" strings take a bit more working with, but the benefits from a memory management point of view are worth the time taken to learn about working with them.

josef2600, 2023-03-21:
i am working on this too:
PreferenceHandler.cpp
`    else if (strcmp(preference, PREFERENCES_AUTOMATION) == 0) 
    {
        // Note: saving a string instead of bytes. 
        // The struct for AutomationFlash does not seem to work properly with the putBytes. 
        // Need further investigation
        preferences.putString(PREFERENCES_AUTOMATION, getAutomationsJson());
`
i changed it (and more!) to this:

`        getAutomationsJson();
        preferences.putBytes(PREFERENCES_AUTOMATION, &my_stringbuffer_, sizeof(my_stringbuffer_));
.
.
.
String PreferenceHandler::getAutomationsJson() 
{
    DynamicJsonDocument doc(AUTOMATIONS_JSON_CAPACITY);
    for (AutomationFlash& a : automations) 
    {
        if (a.id) 
        {
            doc.add(serialized(automationToJson(a)));
        }
    }
 //   String output;
    my_stringbuffer_.remove(0);
    serializeJson(doc, my_stringbuffer_);
    return my_stringbuffer_;
}
`

josef2600, 2023-03-21:
for now the only thing i fund is 2 links:
https://www.esp32.com/viewtopic.php?t=29383
https://github.com/espressif/esp-idf/issues/9631
so far no solution. i have to check where is it coming from.

josef2600, 2023-03-23:
i think i found a clue what is what. this project is excellent and works with very little changes, mainly because the library's have been updated and changed some codes. but, the main problem comes from this rep:
`https://github.com/me-no-dev/ESPAsyncWebServer`
and:
`https://github.com/me-no-dev/AsyncTCP`
since these are very very old and extremely boggy codes, everything goes bad. and because i don't work with (or like) C++, and it is extremely hard (and for me and many not possible) to convert it to C, the fix is unlikely. everything is bad! from leaking to just bad codes and misscoding, the ram conflicting to just trow out the SPIFFS and many more. it works some times, because usually  everyone is using a module with PSRAM. but when i have removed it and changed it to only work with one ram and it is low amount, the leaks and mismanaging the registers, makes the CPU to trow bad Exceptions. 
i have tested it with and without using the RTOS too. i ran it without the RTOS and i ran it for 500 times before crash, but the timing is the same. meaning, fast or slow, with or without RTOS in main code (this rap codes with fixes i have said before), in all that, the process works for 1 minute (around it) and crashes. the timing is the same, not how many time this code runs, but how many times the `ESPAsyncWebServer` is running.
and the last fix i have to mention is this:
https://github.com/philbowles/ESPAsyncWebServer/issues/3
Change these three lines in WebAuthentication.cpp:

```
  mbedtls_md5_starts(&_ctx);
  mbedtls_md5_update(&_ctx, data, len);
  mbedtls_md5_finish(&_ctx, _buf);
```
To:
 ```
 mbedtls_md5_starts_ret(&_ctx);
  mbedtls_md5_update_ret(&_ctx, data, len);
  mbedtls_md5_finish_ret(&_ctx, _buf);

```
i don't know what to do except change everything and began from the beginnings. reason is, i don't work with C++ and the code from C++ cannot be converted to C. there are few instructions in C++ that don't Implements in C.
in any case, i have found this repo too. maybe it will help somebody. it didn't help me since it doesn't include all the functions needed. he tried to fix many problems, but didn't do it all.
https://github.com/philbowles/h4plugins
anyway, if i could help in my specialty, please do ask. i will work on this until i am sure i cant do anything more. goodluck to everyone.
best regards, Josef.

RomeHein, 2023-03-23:
Hello @josef2600 
Unfortunately you are right, the intent of ESPecial was to learn basics. It was never made for production.
I encounter the crashes you described. And found out that it was indeed coming from these libraries. Very difficult to debug.
If you have a look to other branches of ESPecial repo, you'll find an attempt of using H4 (the library you are suggesting). I think this is the right way to go, but this would require a full rewrite of ESPecial. I don't have time at the moment for this.

I'll start soon a new esp32 project, and I will for sure follow your advice about C/C++ and I'll go the H4 road as it will give the right tooling without spending too much time on learning.

josef2600, 2023-03-24:
thank you @RomeHein , i have been working on automation projects for 18 years. i have a patented invention on one. but until 2 years ago, i was always working with PIC (microchip) for them. my last project was around 18 month ago. it was beautiful and wonderful. the only problem was, i used a PIC32MZ2048EFH144 for it. and the price for that thing was 25$ alone! it had a huge PCB and a lot of expensive other parts. including an extra 32MB DDR2 ram on it. in total, the price got to around 90$ just for one board. and the programming for that was enormous! in general, for everyone using, it was too expensive. so i was looking and searching. at least i checked 500 projects from cheap micro to using a PC! and among all of them, the best one that i loved, was yours. your idea was excellent and with a cheap module. the reason i chose the ESP32-C3 is simple. it wasn't the price, but the specs. i chose `ESP32-C3FN4` because it can handle temp up to 105 °C. the next best is S3 with internal 2MB internal psram, but only goes up to 85°C. both have 4MB internal flash. i designed my own antenna for them and it is excellent. this is my work:
[https://drive.google.com/file/d/1q3-kTbD4nNodlmIjtXle3ac3R4p9MVp4/view?usp=sharing](url)
[https://drive.google.com/file/d/1rjRe8KgQiB1MUPzGF67mWDhQ5aRISkvK/view?usp=sharing](url)

ok, back to program. i found that the async lib has a big problem (among 50 more!). as i said, it uses the RTOS, the big big problem with this kind of working is when you use it, you cant know when 2 or more functions are going to clash. for example in this particular work, it is crashing because `events.send` are going simultaneously to output. and it is bad! if i was the programmer, the solution would have been simple, but i am not! neither are you. so, i am working on it to go to H4 lib. but it doesn't include all of the parts your project needs. lets see if i can go farther or not! 
the thing i am trying to say is, thank you, but if you can, continue your work. 
the other thing i am confused about in your code is this:
in file `esp32.ino` there are 2 void with the same name, why? am i missing something?
`void runAutomation(int id)` and `void runAutomation(AutomationFlash& automation)`
thank you. best regards, Josef.

RomeHein, 2023-03-24:
> void runAutomation(int id) and void runAutomation(AutomationFlash& automation)

This is known as function overloading, which is a feature of the C++ language that allows you to define multiple functions with the same name but different parameter types or numbers. But I guess you are horrified by this type of things 😅
In general, I would say it is good practice to use descriptive function names to avoid confusion when overloading functions, especially if they have similar functionality. So yeah, maybe change the name, but be aware to change the functions accordingly in the code 😉

At the moment I only have classic esp32, but also esp32-s2 for low power projects. I bought these only for the price which is super low, and the form factor super small. I'll go the C3 or S3 road once the price will be lower.

By the way, I have changed the telegram channel to be open for discussion: https://t.me/especial32
You can join there and we can discuss all this ;)

josef2600, 2023-03-24:
oh man, great! and yes! i was going crazy how is it working! 
but i have a fantastic news for you! i will try telegram again, hope works! only there i am going to send you the full work. right now i am so tired i can do anything and as i have already pissed you off(!) i wont send the files here. well, i don't know how! 
any way, that fff H4 man! that repo and libs work! i cant believe it! with those libs and few fixes that i already said and 1 more fix, it can run perfectly fine. the last fix is to convert `serverhandler->events.send` to a `static void` and no hangs, and no leaks any more. i have tested it with 2 user and for over 1 hour. just last thing you have to know is this, all this sh%&y chips have problem with USB! again! just dont use them or it will leak so much that memory will blow up! and it freeze too. dont use USB for final work. this is the link:
https://github.com/espressif/arduino-esp32/issues/6089
and thank you. hope we can work more or be friends. i like your mind. it is a rare thing! i assure you! just it takes time to get use to me!
and please, for microcontrollers, any micro, the C++ is a no no no no no!

closed 2023-03-25 by josef2600