RomeHein/ESPecial #24: 'return' in middle of 'VOID' is a no no

josef2600, 2023-03-21:
let me make it simple, NEVER use return in void. this can cause many problems.  void keyword specifies that the function doesn't return a value. and, beside that, depending on the compiler, it may even not clear the registers in it. but worth of all, if the jumps gets too many, it causes crash. each CPU has a finite and usually low amount of stack. if the stack gets full, the CPU usually hangs and might cause unsuitability or reset. it causes many memory leaks too. instead of `return` use `else` or `else if`.

josef2600, 2023-03-21:
well, just sow this:
in `ESPecial.ino`
in `void execOTA(const char* version){`
in `switch` you used `return` instead of `break` . that's new! i am fixing it.

josef2600, 2023-03-21:
another example:
this:

void ServerHandler::handleAutomationEdit(AsyncWebServerRequest *request,JsonVariant &json)
{
    JsonObject doc = json.as<JsonObject>();
    for (AutomationFlash& a : preference.automations)
    {
        if (a.id == doc["id"].as<int>())
        {
            int16_t conditions[MAX_AUTOMATIONS_CONDITIONS_NUMBER][4] = {};
            int j = 0;
            for(JsonArray condition: doc["settings"]["conditions"].as<JsonArray>()) 
            {
                int16_t conditionToFlash[4];
                for (int k = 0; k<4; k++) 
                {
                    conditionToFlash[k] = condition[k];
                }
                memcpy(conditions[j], conditionToFlash, sizeof(conditionToFlash));
                j++;
            }
            char actions[MAX_AUTOMATION_ACTION_NUMBER][4][100] = {};
            int l = 0;
            for(JsonArray action: doc["settings"]["actions"].as<JsonArray>()) 
            {
                char actionToFlash[4][100];
                for (int k = 0; k<4; k++) 
                {
                    strcpy(actionToFlash[k], action[k].as<const char*>());
                }
                memcpy(actions[l], actionToFlash, sizeof(actionToFlash));
                l++;
            }
            request->send(200, "text/json", preference.editAutomation(a, doc["settings"]["label"].as<const char*>(),doc["settings"]["autoRun"].as<int>(),conditions,actions,doc["settings"]["loopCount"].as<int>(),doc["settings"]["debounceDelay"].as<int>()));
           return;
        }
    }
 request->send(404, "text/plain", "Not found");
}

===========================
will be converted to:

void ServerHandler::handleAutomationEdit(AsyncWebServerRequest *request,JsonVariant &json)
{
  bool cchek = true;
  
    JsonObject doc = json.as<JsonObject>();
    for (AutomationFlash& a : preference.automations)
    {
        if (a.id == doc["id"].as<int>())
        {
            int16_t conditions[MAX_AUTOMATIONS_CONDITIONS_NUMBER][4] = {};
            int j = 0;
            for(JsonArray condition: doc["settings"]["conditions"].as<JsonArray>()) 
            {
                int16_t conditionToFlash[4];
                for (int k = 0; k<4; k++) 
                {
                    conditionToFlash[k] = condition[k];
                }
                memcpy(conditions[j], conditionToFlash, sizeof(conditionToFlash));
                j++;
            }
            char actions[MAX_AUTOMATION_ACTION_NUMBER][4][100] = {};
            int l = 0;
            for(JsonArray action: doc["settings"]["actions"].as<JsonArray>()) 
            {
                char actionToFlash[4][100];
                for (int k = 0; k<4; k++) 
                {
                    strcpy(actionToFlash[k], action[k].as<const char*>());
                }
                memcpy(actions[l], actionToFlash, sizeof(actionToFlash));
                l++;
            }
            request->send(200, "text/json", preference.editAutomation(a, doc["settings"]["label"].as<const char*>(),doc["settings"]["autoRun"].as<int>(),conditions,actions,doc["settings"]["loopCount"].as<int>(),doc["settings"]["debounceDelay"].as<int>()));
           // return;
           cchek = false;
           break;
        }
    }
    if (cchek) request->send(404, "text/plain", "Not found");
}
=================================
now `return` doesn't exist and code is functioning properly.

RomeHein, 2023-03-21:
I'm sorry but your comments/issues are really difficult to follow.
Please please please open PRs, format your code, structure your thoughts, deal issues one by one 🙏🙏🙏 

Otherwise your work won't get merged to this project, that would be a pretty huge waste of time for both of us.
It seems that you have some knowledge of programming language, that's good. But you absolutely need to gain some about how things work on Github ! 

⚠️ This is a mandatory things before going forward. Please have a look on internet, they are tons of tutorials: "how to contribute to github open source projects" => https://www.dataschool.io/how-to-contribute-on-github/

closed 2023-07-01 by josef2600