RomeHein/ESPecial #26: convert SPIFFS to LittleFS

josef2600, 2023-04-05:
the `SPIFFS` has been out of support for a long time and has many problems. you can change it to `LittleFS` easily. 
1. replace all `SPIFFS` with `LittleFS` . and replace all `#include <SPIFFS.h>` to `#include <LittleFS.h>` . 
2.  you can do the same for `FFat` too. 
3. the only change in code is `freeBytes()` . for this you can do this `LittleFS.totalBytes() - LittleFS.usedBytes()` .
and that is that!
the speed is 4 to 5 times faster to `LittleFS` . and it is resistance to many problems, included power lost and corruption of files. and probably to badsectors as well. and the amount of used ram is very low. specially compare to FAT, it is nothing! fat uses more than 45 KB more and when the page is open, it uses tow time the space in ram per file. in general it is better than fat, except in speed. it doesn't make a lot of different though! it is microcontroller anyway!

josef2600, 2023-04-05:
and for partition scheme, use any `SPIFFS` partition for project, then use app for downloading `LittleFS` to esp. the app link:
https://github.com/lorol/arduino-esp32fs-plugin

RomeHein, 2023-04-10:
> the `SPIFFS` has been out of support for a long time and has many problems. you can change it to `LittleFS` easily. 
> 
> 1. replace all `SPIFFS` with `LittleFS` . and replace all `#include <SPIFFS.h>` to `#include <LittleFS.h>` . 
> 
> 2.  you can do the same for `FFat` too. 
> 
> 3. the only change in code is `freeBytes()` . for this you can do this `LittleFS.totalBytes() - LittleFS.usedBytes()` .
> 
> and that is that!
> 
> the speed is 4 to 5 times faster to `LittleFS` . and it is resistance to many problems, included power lost and corruption of files. and probably to badsectors as well. and the amount of used ram is very low. specially compare to FAT, it is nothing! fat uses more than 45 KB more and when the page is open, it uses tow time the space in ram per file. in general it is better than fat, except in speed. it doesn't make a lot of different though! it is microcontroller anyway! 
> 
> `LittleFS ` is included in the ESP Arduino boards and doesn't need any extra lib. it is updates with board manager.

That looks like a really good and easy enough improvement. Could you open a PR with only the fix to support LittleFS ?

lyxer123, 2023-06-24:
I have solve this problem，and I port this project to platformio system.
link:  https://github.com/lyxer123/ESP32-Automation

closed 2023-07-01 by josef2600